smeisner/smart-thermostat #7: Local Weather

smeisner, 2023-04-01:
Research usefulness and ability to fetch weather & temp for the local area (via something like wunderground.com). The local weather can then be used to set the background graphics on the touch screen.

rprobaina, 2024-06-05:
Hi @smeisner,

I was talking to @aquini about this amazing project yesterday, and I realized that for this issue https://wttr.in/nashua?format=j1 might be a good fit for a quick experimentation, since an API key isn't required.

https://github.com/chubin/wttr.in

smeisner, 2024-06-05:
This is great. I was also thinking about how to supply the user's location, but looking at geoip services (like https://www.maxmind.com/en/locate-my-ip-address) that may be fairly easy.

Any interest in taking on this issue?