smeisner/smart-thermostat #19: Temperature via matter

beckmx, 2023-09-04:
Hello there, I was looking at your code and was wondering if you managed to set the temperature into the matter protocol, I just noticed this part **local_temperature**, or do you have any clue on how to send the temperature via matter?

smeisner, 2023-09-04:
This PCB & code are almost ready for "V1", but it is not there quite yet. Actually on another branch I have been rewriting all of the code to NOT use the Arduino framework and only use ESP-IDF. Once I get that code committed, I will be returning to the Matter implementation.

As far as Matter, so far I have managed to get Matter up and running with the CHIP ESP component and have the server (HomeAssistant) recognize my thermostat. Doing just that was quite an exercise. The very next step is to allow HomeAssistant to control the mode and temp using Matter over wifi ...and to report back to HA the current states.

So, to answer your question, no, not yet.  :D I believe sending the temp back is done via a Matter callback, but I need to investigate further.

beckmx, 2023-09-06:
use this, i tried it and joined ur code with this, worked nicely https://github.com/aleksigold/esp-matter-temperature

closed 2023-09-06 by beckmx