smeisner/smart-thermostat #28: Turn on fan pin with heat or cool

smeisner, 2023-12-02:
(G) Fan must be energized whenever (W) HEAT or (Y) COOL is asserted.

smeisner, 2023-12-02:
Many modern HAVAC systems do not require this, but older ones can very well require FAN when heat or cool is called for.

y9rabbito, 2023-12-02:
Can you specifically elaborate it and in which file i need to update the fan control signal G.

smeisner, 2023-12-02:
the relay pins are controlled in state_machine.cpp. In the main loop is where the decisions are made to turn on/off each relay...and LEDs, if appropriate.

y9rabbito, 2023-12-02:
@smeisner means this value gpio_set_level((gpio_num_t)LED_FAN_PIN, LOW); should be set to high on coolstate currenttemp > maxtemp

smeisner, 2023-12-02:
Yes, that will turn on/off the LED. I think you want to look for HVAC_FAN_PIN to control the relay. For heat, the red LED is turned on. For cool, the blue is turned on. If only the fan is turned on, the green LED is turned on. If the fan is turned on as part of cooling/heating, there is no need to turn on the LED.

y9rabbito, 2023-12-02:
Actually i have never read or know about HVAC system Im googling the topic and trying to understand the code.

smeisner, 2023-12-04:
This issue was addressed with e7d7b29dbfe0932cfff6a9a89d79f6e1776096d0. 
Code has beenmerged.

closed 2023-12-04 by smeisner