smeisner/smart-thermostat #49: Versioning of software & hardware

DSpeichert, 2024-07-22:
It seems that hardware has versions like v0.5.2 and v0.6.0 (latest), while software/firmware has gotten up to v0.7.3.

At the same time, the tagged v0.7.3 will have PCB/kicad designs of prior versions.

This seems unnecessarily confusing. It does make sense to version firmware & hardware separately (along with breaking changes/incompatibilities between them), is there a better way to do that?

smeisner, 2024-07-23:
Yes...the rate of change between the PCB and the firmware is completely different. HW and FW will always have different version numbers.

The issue with the PCB CAD files is related to the long development & test time for each PCB. While working in KiCad, I do not know if the mods I am making will be worthy of a new rev number until I place an order, wait for manufacturing, shipping and then populating the board and finally testing it. That could easily be 2 - 3 months of time.

The version numbering of the KiCad files will align with current releases once I get passed the beta phase. I would like to set the FW and PCB @ V1.0.0 upon "public" release.

smeisner, 2024-07-23:
Actually, the more I think about this, the more I realize I want to have a versioning scheme that will make it obvious what FW versions will work on what PCB versions. For ex, if there is something new in version 2.0 of the PCB, then only FW versions that start with V2.x.x will work on that HW. Conversely, FW V2.x.x will work on any V2.x version of the PCB. If there is a breaking change, major version numbers must change as well.

Does that work?

jrtaylor71, 2025-01-27:
Is there software for version 0.6.0 hardware?

smeisner, 2025-01-27:
> Is there software for version 0.6.0 hardware?

There is, but do you actually have a v0.6.0 prototype?

I will need to look up the details of that HW, but I believe the latest sw will work on 0.6.0.

jrtaylor71, 2025-01-28:
Yes I had several made as it was cheaper. I also have hardware to assemble a few. The v0.6.0 hardware is going to be hard for the average person to assemble as some of the small parts require hot air assemble with a microscope.

I am a retired communications engineer so this kind of stuff was part of my day to day job just not programming. I can help with the hardware for this project. I have limited C coding abilities but from what I can tell the pin numbers used on hardware 0.4.0 and 0.6.0 are different.

smeisner, 2025-01-30:
> Yes I had several made as it was cheaper. I also have hardware to assemble a few. The v0.6.0 hardware is going to be hard for the average person to assemble as some of the small parts require hot air assemble with a microscope.

After going thru all of the historical data, I have created a sheet detailing the use & change of GPIO pins. As of V0.3/0.4, the pins have not changed much to V0.6 of the HW:
- SWD interface was removed (IO 39, 40, 41, 42) <-- it was not useful
- LD2410 Serial interface was added (IO 15& 16) <-- LD2410 used for human detection
- General purpose UART was added (IO 36 & 37) <-- Used for serial logging
- USB interface was added (IO 19, 20) <-- Use the D1 Mini USB interface

If I recall correctly, the version you have has the Lolin D1 Mini on it, right?

If all my research is accurate (the details above), then using the latest firmware will still work. You won't get person detection as you approach the unit (to bring up the touch screen backlight) and you need to use the D1 Mini USB interface for programming & data logging.

Let me know if I have things wrong or if you need help.

Steve

> 
> I am a retired communications engineer so this kind of stuff was part of my day to day job just not programming. I can help with the hardware for this project. I have limited C coding abilities but from what I can tell the pin numbers used on hardware 0.4.0 and 0.6.0 are different.

jrtaylor71, 2025-02-01:
> After going thru all of the historical data, I have created a sheet detailing the use & change of GPIO pins. As of V0.3/0.4, the pins have not changed much to V0.6 of the HW:
>
>     * SWD interface [...]

No it uses the with a USB_C_Receptacle_USB2.0. 

> If all my research is accurate (the details above), then using the latest firmware will still work. You won't get person detection as you approach the unit (to bring up the touch screen backlight) [...]

[image]
[image]

smeisner, 2025-02-01:
.
> > 
> > After going thru all of the historical data, I have created a sheet detailing the use & change of GPIO pins. As of V0.3/0.4, the pins have not changed much to V0.6 of the HW:

> > If I recall correctly, the version you have has the Lolin D1 Mini on it, right?
> 
> No, it uses the ESP32-S3-WROOM-1-N16 with a USB_C_Receptacle_USB2.0. 

The pictures show v0.6.0. Based on those pics, I can tell you you have the latest released hardware. The pin definitions in the firmware will work. 

Steve

jrtaylor71, 2025-02-01:
OK I will let you know when I finish getting one put together.

jrtaylor71, 2025-02-20:
I have not have time to get a board put together so I just ordered some more and them put together. The BOM is missing a few components like relays, buzzer and headers are not a big deal but BR1, r6, r7 and r12 is missing.

smeisner, 2025-02-20:
Yes, completely intentional. These are either components JLCPCB could not source, or were thru-hole components I could solder in later and cheaper.

jrtaylor71, 2025-02-20:
Could you list the headers and buzzer that you planned? I have the other parts.

smeisner, 2025-02-20:
A lot more than you asked for, but I supplied everything I see missing in your photo.

- BZ1: TDK PS1240P02BT
-- https://www.digikey.com/en/products/detail/tdk-corporation/PS1240P02BT/935924
- K1 - K5 Relays: Kemet 551-UC2-3NJ
-- https://content.kemet.com/datasheets/KEM_R7005_UC2_UD2.pdf
-- I believe Mouser had the best price on these
- BR1: MB6S 
-- https://www.smc-diodes.com/propdf/ABS22%20THRU%20ABS210%20N1924%20REV.A.pdf
- R12: GL5537 / LDR07
-- http://www.tme.eu/de/Document/f2e3ad76a925811312d226c31da4cd7e/LDR07.pdf
-- https://www.amazon.com/gp/product/B01M3TO7RA
- J3: Phoenix Screw Terminal Block
-- https://www.digikey.com/en/products/detail/on-shore-technology-inc/OSTVN10A150/1588870
- The LD2410 motion sensor:
-- https://www.aliexpress.us/item/3256804932047824.html?spm=a2g0o.order_list.order_list_main.10.41951802dzj23w&gatewayAdapt=glo2usa&_randl_shipto=US

J2, J5, J6 are just the common 2.54 pitch pin headers (For J6 I use a right angle one and point them downward)

smeisner, 2025-02-20:
And just so you know...the "JLCJLCJLCJLC" is a string used by JLCPCB as an indicator where to put the production number. You need to select "Order Number (Specify Position)":

[image]

jrtaylor71, 2025-02-20:
Thanks, I have a touch screen. Thanks for the JLCPCB info

jrtaylor71, 2025-02-20:
This is the I got.

DIANN 3.2" ILI9341 SPI TFT LCD Display Touch Panel 320x240 TFT LCD Touch Screen Shield 5V/3.3V STM32 Display Module SPI Serial with Touch Pen
https://a.co/d/9GtUCAT

smeisner, 2025-02-20:
Yeah, that's the same one I am using. Originally when I was first starting out, I got this one:
https://www.amazon.com/dp/B0B1M9S9V6

The ones from aliexpress are the same...just much cheaper.

jrtaylor71, 2025-02-25:
What is the CLI command to write the firmware to flash without building it? If I build the current 0.7.4 there is no wifi scanning. It just says no-ssid.

smeisner, 2025-02-26:
If you want to fix this yourself, swap lines 151 & 152 in wifi.c :

```
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));

```
to 

```
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));

```
The side effects of the `esp_wifi_scan_get_ap_num()` call was changed to reset the number of APs found. Putting it first maintains the count of APs found.

This discussion should be under its own issue. We have successfully hijacked the original intent of this issue.

smeisner, 2025-02-26:
Issue #63 has been created. Please continue discussion there.

smeisner, 2025-03-19:
Moving this Issue to the Discussions section.