smeisner/smart-thermostat #50: Cold boot into bootloader

DSpeichert, 2024-07-23:
From GettingStarted.md:

> NB #3: If the board has been powered off for a bit, it starts up in boot mode when power is applied. Therefore, the reset button must be pushed. I am guessing this is a power-up problem...if anyone can suggest a PCB change, I will include it in the next run!

Potentially related:
* https://esp32.com/viewtopic.php?t=14147
* https://github.com/espressif/esp-idf/issues/11448

jrtaylor71, 2025-03-20:
Per the data sheet that is incorrect. See pdf pages 29-30 and 45.

https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf

**By default, GPIO0, GPIO45, and GPIO46 are connected to the chip’s internal pull-up/pull-down resistors.** See pdf page 534.

https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#bootctrl

Not trying to argue but I have 30+ of doing this.

smeisner, 2025-03-20:
Two things to consider: 
1. This unit is still a prototype, as indicated in the silkscreen of the PCB
2. A thermostat device is an "always-on" device. Power-up is not a primary concern right now, especially with a simple workaround (press the reset button). HVAC features and reliable communications is the current primary concern.

Another point to mention, you stated you figured it out (remove the capacitor).

I too am not "trying to argue" especially since I have not yet researched this topic fully. But, I am going to set priorities as I see fit and right now developing the firmware is a higher priority.

smeisner, 2025-03-21:
Thanks for the feedback on the board design. I hadn't had anyone review it yet.

I did spend some time reading the docs you linked and can tell you are correct that capacitor was the cause of the problem. My prior comment regarding the pull up resistor was my "drive by" solution I had ...I had not yet done the analysis and proper review of the idea. I was quick to assume IO0 was left floating and causing the issue...which wasn't the case at all.

My updated PCB design has already removed C7.

jrtaylor71, 2025-03-21:
The only other thing that I saw was that the internal layers are a little close to the threw holes for mounting. We use to keep them outside the expected screw or mounting post outer diameter size.

smeisner, 2025-04-16:
Latest prototype removes the "debounce capacitor" (C7 on V0.6) and problem is gone.

closed 2025-04-16 by smeisner