smeisner/smart-thermostat #51: Is UART/Serial port redundant?

DSpeichert, 2024-07-23:
Apparently, the USB-Serial-JTAG is unique to ESP-S3/C3 and effectively provides power, firmware upload and logging and thus it can send logs unless in deep-sleep mode (or disconnected).

Do we ever expect to need two serial interfaces for logging? Or is one specifically to be used for REPL?

[USB-SERIAL-JTAG docs](https://docs.espressif.com/projects/esp-iot-solution/en/latest/usb/usb_overview/usb_serial_jtag.html)

smeisner, 2024-07-28:
> Apparently, the USB-Serial-JTAG is unique to ESP-S3/C3 and effectively provides power, firmware upload and logging and thus it can send logs unless in deep-sleep mode (or disconnected).
>
> Do we ever [...]

Yes, it is unique to the S3 and is utilized by this prototype for FW upload and debugging. The other UART built into the S3 is used to program the LD2410 presence sensor. As described elsewhere, currently this is a prototype and the use of these 2 UARTs is redundant on purpose.

closed 2024-07-28 by smeisner

smeisner, 2024-07-28:
Also, please refer to issue #45  for future use of UART port