smeisner/smart-thermostat #52: Battery backup to avoid rapid cycling when power flickers

DSpeichert, 2024-07-28:
Battery backup could help avoid dangerously frequent cycles during flickering power (commonly recommended is minimum 2 minutes of on/off time for AC).

Alternatively, a mandated 2 minute "off" time upon power up could be enforced in firmware.

smeisner, 2024-07-28:
Once again...this is a prototype. There will be no battery backup added to this design. The cost is already is a bit much for this unit and we need ways to lower it. All of the relevant data is stored n NVRAM so there is no need for battery backup as of yet.

The idea of a delay is fine, but 2 minutes is useless. What I've been playing with is more like 15 seconds. Today's HVAC systems already take care of what you're referring to ... and thermostats today (many of the so-called smart ones) don't have any built in delay at all.

I am also investigating delays after changing the target temp. In other words, once the user sets a target temp, wait 2 seconds before commiting the change. This is useful while the user stands there thinking if that's the right temp they actually want.

smeisner, 2024-07-28:
I also have a super capacitor (as mentioned in one of the intro docs) to see if that would work as a short-term "battery" backup. Again, due to the cost, I am not sure it's worth it.

closed 2024-07-28 by smeisner