smeisner/smart-thermostat #54: 24VAC power tolerance

DSpeichert, 2024-07-28:
The HVAC's "24VAC" in the wild could measure up to 27-28VAC.

Honeywell T4 thermostat has [tolerance for 20-30 VAC](https://digitalassets.resideo.com/damroot/Original/10014/33-00614EFS.pdf):
[image]

smeisner, 2024-12-17:
@DSpeichert Were you citing a problem here? I believe the thermostat may handle this range of input voltages. The LM2596T-5 can handle up to 45V input and the MB6S rectifier can handle even more. So is there a problem you are seeing?

DSpeichert, 2024-12-30:
I honestly don't remember anymore, I was likely not aware of the actual tolerance currently supported by the board. This is mostly an information/documentation thing to make keep some tolerance in the design (especially if you plan to simplify the power handling).

smeisner, 2025-02-27:
Closing due to inactivity. Feel free to reopen if this is an issue.

closed 2025-02-27 by smeisner