smeisner/smart-thermostat #62: Are the 3v3 and GND layers actually necessary?

DieKatzchen, 2024-12-16:
They cause the manufacturing cost to skyrocket. a 101.5mm x 83.4mm four layer board costs me ~$140, a two-layer instead costs ~$25. If 1.5mm could be trimmed off the long edge it would bring it down to ~$5.

smeisner, 2024-12-16:
I have been using JLCPCB and I see nothing costing this much. Which PCB manufacturer are you using?

Are the 2 layers necessary? No. But since this is a prototype and routing is already tricky, it facilitates the firmware development. Feel free to email me directly at: steve at meisners dot net and we can discuss other arrangements if you want.

DieKatzchen, 2024-12-17:
Huh, I didn't realize JLCPCB had such reasonable 4-layer pricing. I was using PCBWAY. Now the only problem is that they have a shortfall of some of the 0603 resistors and I really don't feel like hand-soldering 0603s. I suppose there's no rush.

smeisner, 2024-12-17:
If you are interested in **contributing** to this project, I do have a thermostat (fully assembled and tested) I can send you. If interested, let me know.

DieKatzchen, 2024-12-17:
What kind of contribution do you need? I'm not particularly good at hardware development, but I'm decent at software. If you're using Platformio I've used it before, on Espressif chips even.

smeisner, 2024-12-17:
> What kind of contribution do you need? I'm not particularly good at hardware development, but I'm decent at software. If you're using Platformio I've used it before, on Espressif chips even.

Yes, the project (under the "App" folder) is fully PlatformIO based and ESP-IDF based. Take a look at the Issues tab here on Github and at the code and let me know what you think.

DieKatzchen, 2024-12-18:
I'll take a look later. Currently really busy, but hopefully less so soon.

smeisner, 2024-12-18:
Sure. Let me know if you need any help.

smeisner, 2025-02-26:
Closing due to lack of activity. Feel free to reopen in desired.

closed 2025-02-26 by smeisner