smeisner/smart-thermostat #63: No SSIDs found during scan

smeisner, 2025-02-26:
When performing a wifi scan, no APs are found. This is due to a problem with an ESP SDK call that changed its side effects. Lines 151 & 152 in wifi.cpp must be swapped to maintain the count of APs found. As it is now, the count is always reset to 0.

```
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));

```

Change the above lines to:
```
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));
ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));

```
The side effects of the `esp_wifi_scan_get_ap_num()` call was changed to reset the number of APs found. Putting it first maintains the count of APs found.

jrtaylor71, 2025-02-26:
Wifi scan works with this change.

smeisner, 2025-02-26:
Fixed with commit https://github.com/smeisner/smart-thermostat/pull/67

closed 2025-02-26 by smeisner