smeisner/smart-thermostat #64: timezone off by 12 hours

jrtaylor71, 2025-02-26:
The timezone setting is off by 12 hours. I am in the US central timezone which is -6 or -5 DST. To get the correct time I needed to set it to +6.

smeisner, 2025-02-26:
The latest version of the code includes a function to automatically set your timezone based on the GeoIP information. Are you not relying on that and setting it manually?

Here in New Hampshire (Eastern time), I set it to GMT -5 and it works fine...if I set it manually and not rely on the GeoIP function.

jrtaylor71, 2025-02-26:
Not sure what happened. I set it to +6 and it has the correct time. Then reset it and now it shows -6 with the correct time. There is no automatic option.

[image]

smeisner, 2025-02-27:
The 'Automatic option' is not an option. It is code in the firmware to lookup the Geo location based on the IP address and set the timezone. If that fails, then it falls back to whatever you set and save via the screen menu.

smeisner, 2025-03-08:
@jrtaylor71 After re-reading your last comment above, this makes sense. Based on the GeoIP information, you are at GMT-6 not +6. When the code to determine your location and timezone ran, it updated your location to -6.

It's very possible while you were setting it manually it was updating in the background. This could have lead to the perception you needed to set it to an odd timezone value.

smeisner, 2025-03-18:
Closed due to inactivity.

closed 2025-03-18 by smeisner