smeisner/smart-thermostat #65: Addition of external temp sensor

jrtaylor71, 2025-02-26:
I would be nice to see an an option for an external temp sensor like a DS18B20 for hydronic heating and outdoor boiler applications. This would allow the fans to be shutdown or internal pumps when the incoming water temp gets below a certain point. A typical heating system like this runs between 150 and 180 degF. This way you are not pulling heat from the building you are trying to heat. I would add this to the web interface and not the display.

smeisner, 2025-02-26:
This is a good idea. What I originally had in mind was utilizing a sensor (maybe via HomeAssistant) that supplied the temperature. What you're suggesting is a hardware interface for a local temp sensor? Maybe adding a screw terminal for the 18B20?

jrtaylor71, 2025-02-26:
18B20 or similar would be best as they could be installed in a temperature well. This way they get the best reading. I built a monitor for my boiler that uses a kc868-am with 18b20 sensors and they get the same reading as the gauge on the boiler.

[image]

smeisner, 2025-04-07:
@jrtaylor71 The board has 3 GPIO (33, 34 and 35) pins broken out and are available for general use. Can these be used for what you are thinking?

[image]