smeisner/smart-thermostat #66: mqtt not working

jrtaylor71, 2025-02-26:
No mqtt connection atempt to home assistant. I tried with and without port number at the end of the broker address. I will post debug info when I have the serial port connected.

smeisner, 2025-02-26:
Can you also telnet to the thermostat and issue a "status" command. Please provide that info as well.

jrtaylor71, 2025-02-26:
Settings are randomly being lost when using the reset button. It is connected now but when I first looked at the mqtt was enabled but status was disabled. Then I reset it and the display showed it was disabled. So I enabled it and then that when it connected.

The same thing is happening with the cool and fan setting.

`Current Status:
--------------------------------------------------------
Firmware version: 0.7.4
Firmware build date: Feb 26 2025 - 07:45:58
Device name: thermostat
Friendly name: Thermostat
Uptime: 0 days, 00:00:16
Free Heap size: 142496
MAC Address: f0:9e:9e:54:d3:74
Wifi SSID: jonn
Wifi connected: Yes
Wifi signal: 100%
Wifi IP address: 192.168.183.104
Current time: 13:57:08    Timezone: GMT-6
MQTT Enabled: Yes
MQTT Connected: Yes
MQTT Broker: 192.168.183.238  Port: 1883
MQTT Username:  mqtt2
MQTT Password:  mqttxxxxxxx
Current temp: 79.7 F (Correction: -4.8)
Current humidity: 40.9% (Correction: 10.0)
Target temp: 69.0 
Swing temp: 1.5 
Light detected: 1594
Motion detected: No
Display sleep time: 70
Touch screen beep: Enabled
Current HVAC mode set: Off (Currently: Off)
HVAC modes enabled:
  Heat: True  Cool: False  Fan: False
  2-stage heating enabled: False
  Reversing valve (heat pumps): False
`

`
Current Status:
--------------------------------------------------------
Firmware version: 0.7.4
Firmware build date: Feb 26 2025 - 07:45:58
Device name: thermostat
Friendly name: Thermostat
Uptime: 0 days, 00:01:12
Free Heap size: 139932
MAC Address: f0:9e:9e:54:d3:74
Wifi SSID: jonn
Wifi connected: Yes
Wifi signal: 100%
Wifi IP address: 192.168.183.104
Current time: 13:58:58    Timezone: GMT-6
MQTT Enabled: Yes
MQTT Connected: Yes
MQTT Broker: 192.168.183.238  Port: 1883
MQTT Username:  mqtt2
MQTT Password:  mqttxxxxxxx
Current temp: 79.8 F (Correction: -4.8)
Current humidity: 41.4% (Correction: 10.0)
Target temp: 70.0 
Swing temp: 1.5 
Light detected: 1553
Motion detected: No
Display sleep time: 70
Touch screen beep: Enabled
Current HVAC mode set: Off (Currently: Off)
HVAC modes enabled:
  Heat: True  Cool: False  Fan: False
  2-stage heating enabled: False
  Reversing valve (heat pumps): False
`

jrtaylor71, 2025-02-27:
See below... All i did was press the reset button.

`Current Status:
--------------------------------------------------------
Firmware version: 0.7.4
Firmware build date: Feb 26 2025 - 07:45:58
Device name: thermostat
Friendly name: Thermostat
Uptime: 0 days, 21:59:17
Free Heap size: 145072
MAC Address: f0:9e:9e:54:d3:74
Wifi SSID: jonn
Wifi connected: Yes
Wifi signal: 100%
Wifi IP address: 192.168.183.104
Current time: 11:57:04    Timezone: GMT-6
MQTT Enabled: Yes
MQTT Connected: Yes
MQTT Broker: 192.168.183.238  Port: 1883
MQTT Username:  mqtt2
MQTT Password:  mqtt652479
Current temp: 77.9 F (Correction: -4.8)
Current humidity: 38.0% (Correction: 10.0)
Target temp: 70.0 
Swing temp: 1.5 
Light detected: 1976
Motion detected: No
Display sleep time: 70
Touch screen beep: Enabled
Current HVAC mode set: Off (Currently: Off)
HVAC modes enabled:
  Heat: True  Cool: True  Fan: True
  2-stage heating enabled: False
  Reversing valve (heat pumps): False
`
`Current Status:
--------------------------------------------------------
Firmware version: 0.7.4
Firmware build date: Feb 26 2025 - 07:45:58
Device name: thermostat
Friendly name: Thermostat
Uptime: 0 days, 00:00:15
Free Heap size: 141836
MAC Address: f0:9e:9e:54:d3:74
Wifi SSID: jonn
Wifi connected: Yes
Wifi signal: 100%
Wifi IP address: 192.168.183.104
Current time: 11:57:28    Timezone: GMT-6
MQTT Enabled: Yes
MQTT Connected: Yes
MQTT Broker: 192.168.183.238  Port: 1883
MQTT Username:  mqtt2
MQTT Password:  mqtt652479
Current temp: 78.3 F (Correction: -4.8)
Current humidity: 41.0% (Correction: 10.0)
Target temp: 70.0 
Swing temp: 1.5 
Light detected: 1967
Motion detected: No
Display sleep time: 70
Touch screen beep: Enabled
Current HVAC mode set: Off (Currently: Off)
HVAC modes enabled:
  Heat: True  Cool: False  Fan: False
  2-stage heating enabled: False
  Reversing valve (heat pumps): False
`

smeisner, 2025-03-03:
@jrtaylor71 I have tried what you described multiple times and do not encounter the same problem you are citing. I just wrote the exact steps ([Initial Setup](https://github.com/smeisner/smart-thermostat/blob/main/InitialSetup.md)) I use after a complete erase and start from the beginning.

jrtaylor71, 2025-03-03:
The only I have not tried is to ease and then re-upload.

jrtaylor71, 2025-03-04:
No change. The erase of the esp32-wroom made no difference. It does connect to the mqtt server without a problem. Not sure what changed. Going to close this and open a new issue as what is happening now is a settings save issue on reset.

closed 2025-03-04 by jrtaylor71