smeisner/smart-thermostat #68: Fan schedule

jrtaylor71, 2025-02-26:
Option to run fan x number of minutes per hour. This allows for air to be circulated through the house when the heat or cool has not ran very often.

yolicap, 2026-09-03:
i was originally pulling from my fork which didnt have the recent tracker changes so i ended up implementing more than i needed to. later i realized that i mostly needed to add a little bit of logic in the `hvacStateUpdate` function.

anyway here are my changes. everything compiles but i cant actually test them rn since my hardware fell apart. ill prob test it over the weekend and create a pr. in the meantime lmk if there's any feedback! 

https://github.com/yolicap/smart-thermostat/commit/c3b14b591426493663f4fb27db65dfaea2692f88

smeisner, 2026-09-03:
@yolicap Be sure to pull the latest code...there have been changes in that area!

Do you expect to be doing additional code changes on this project? If so, I can get you a thermostat to use.

yolicap, 2026-09-03:
ill make sure to pull code before opening a pr! 

i would like to contribute some more to this project. im looking to keep my skills in practice since my current job doesnt require much coding. a thermostat would be helpful and appreciated! 

at the moment i'm struggling to provide power to my setup. i dont think my laptop can provide enough voltage to turn on the screen. i am trying to supply 5vs through a barrel jack but it's a work in progress

smeisner, 2026-09-03:
Shoot me your address at my private email: thermostat@meisners.net

I have a spare set up I can send you. It draws fairly low power via the USB port.

The current list of issues is just the foundation. Next comes some of the fun "smarts" where we can get creative.

yolicap, 2026-09-06:
While testing I found out that the current logic puts the fan in a infinite FAN_ONLY loop. Since we only add tracker values when going from HEAT/COOL/FAN_ONLY to IDLE/OFF. The current runtime never gets updated and stays at 0, therefore the state machine will never exit once it reaches the requested fan runtime. 

I'll make some changes that address this issue on my PR.

yolicap, 2026-09-06:
https://github.com/smeisner/smart-thermostat/pull/106 Is open now. Pls let me know if there are any changes I should make !

smeisner, 2026-09-14:
yolicap has submitted a PR to address this (included in https://github.com/smeisner/smart-thermostat/commit/96a8c14a216ce8c26aa0893cf44421307a66b0b7). It is still a WIP and there will be a UI addition required. But UI changes are blocked by issue #105 .