smeisner/smart-thermostat #74: v0.8.0 going to into download mode on cold power up

jrtaylor71, 2025-03-17:
After latest update it goes into download mode only when you do a cold power up. It does takes like 5 min or after power down to do this. After power up it's fine. I tested this on 5 different units and they all do the same thing.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download

After power up. Either reset button or power cycle.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8920
I (27) boot: ESP-IDF 5.4.0 2nd stage bootloader
I (27) boot: compile time Mar 16 2025 18:22:40
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (71) boot:  2 app0             OTA app          00 10 00010000 00640000
I (78) boot:  3 app1             OTA app          00 11 00650000 00640000
I (84) boot:  4 spiffs           Unknown data     01 82 00c90000 00360000
I (91) boot:  5 coredump         Unknown data     01 03 00ff0000 00010000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=8da68h (580200) map
I (210) esp_image: segment 1: paddr=0009da90 vaddr=3fc9cc00 size=02588h (  9608) load
I (218) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=fc2f0h (1032944) map
I (400) esp_image: segment 3: paddr=0019c318 vaddr=3fc9f188 size=02928h ( 10536) load
I (407) esp_image: segment 4: paddr=0019ec48 vaddr=40374000 size=18b1ch (101148) load
I (430) esp_image: segment 5: paddr=001b776c vaddr=600fe100 size=0001ch (    28) load
I (439) boot: Loaded app from partition at offset 0x10000
I (440) boot: Disabling RNG early entropy source...
I (450) cpu_start: Multicore app
I (459) cpu_start: Pro cpu start user code
I (459) cpu_start: cpu freq: 240000000 Hz
I (459) app_init: Application information:
I (462) app_init: Project name:     espidf-test
I (467) app_init: App version:      V0.8.0
I (472) app_init: Compile time:     Mar 16 2025 18:22:11
I (478) app_init: ELF file SHA256:  e61b10a285274a70...
I (484) app_init: ESP-IDF:          5.4.0
I (489) efuse_init: Min chip rev:     v0.0
I (493) efuse_init: Max chip rev:     v0.99 
I (498) efuse_init: Chip rev:         v0.2
I (503) heap_init: Initializing. RAM available for dynamic allocation:
I (510) heap_init: At 3FCB5148 len 000345C8 (209 KiB): RAM
I (517) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (523) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (529) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (536) spi_flash: detected chip: gd
I (539) spi_flash: flash io: dio
W (544) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (554) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (561) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (569) main_task: Started on CPU0
I (589) main_task: Calling app_main()
E (2059) SENSORS: LD2410: Sensor not connected
I (2069) WIFI: WifiStart()
I (2069) WIFI:   Wifi driver not started; Calling init and create funcs
I (2079) WIFI:   Initializing wifi
I (2079) WIFI:   Setting wifi hostname to: "thermostat"
I (2099) WIFI: MAC Address: f0:9e:9e:54:d3:74
I (2099) WIFI:   Calling esp_wifi_start()
I (2129) WIFI:   Waiting for event callback...
I (4539) WIFI:   event = IP_EVENT_STA_GOT_IP - ip: 192.168.183.104
I (4539) WIFI: Connected to AP SSID: jonn Pre-shared key: Jo****

smeisner, 2025-03-17:
Yes, this issue won't be resolved until I respin the boards. So the fact you are seeing it is expected. It is a PCB hardware issue.

jrtaylor71, 2025-03-17:
What is the problem? Is one of the cpu pins in the wrong state a boot?

jrtaylor71, 2025-03-17:
Did some digging and per https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf you just need to remove C7 across the boot switch.

jrtaylor71, 2025-03-17:
I can confirm that removing C7 fixes the problem.

smeisner, 2025-03-17:
This is a duplicate of #50 . The solution is included in that issue.

smeisner, 2025-03-17:
FYI: C7 is required for debouncing the mechanical switch, The proper fix is a pull up resistor.

closed 2025-03-17 by smeisner

jrtaylor71, 2025-03-18:
Per the data sheet that is incorrect. See pdf pages 29-30 and 45.

https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf

By default, GPIO0, GPIO45, and GPIO46 are connected to the chip’s internal pull-up/pull-down resistors. See pdf page 534.

https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#bootctrl

Not trying to argue but I have 30+ of doing this.