smeisner/smart-thermostat #77: ESP32-S3-Box-3 expansion board?

wrobelda, 2025-03-19:
Have you considered creating an expansion board for ESP32-S3-Box-3 device? This would greatly reduce the complexity of building an ESP32 thermostat and have extra advantage of being able to use the existing functionality of ESP32-S3-Box-3, including support for voice control under ESPHome.

smeisner, 2025-03-19:
No, that's not within the scope of this project. I wanted to create a freestanding thermostat that would replace existing standard units. This means minimizing the external components to create a reliable device. 

Now, having said that, I would absolutely entertain this idea. We could leverage the firmware used here to support this idea.

smeisner, 2025-03-19:
Moving this topic to "Discussions" area.